FreeOpcUa/opcua-client-gui #2: Missing dependency on python3-crypto

mortlind, 2015-12-10:
I got the following error when trying to run v. 0.3.5:

```Shell:
ml@y13:~$ freeopcua-client 
Traceback (most recent call last):
  File "/usr/local/bin/freeopcua-client", line 9, in <module>
    load_entry_point('freeopcua-client==0.3.5', 'console_scripts', 'freeopcua-client')()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 565, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2721, in load_entry_point
    return ep.load()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2377, in load
    return self.resolve()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2383, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/usr/local/lib/python3.4/dist-packages/freeopcuaclient/mainwindow.py", line 10, in <module>
    from freeopcuaclient.uaclient import UaClient
  File "/usr/local/lib/python3.4/dist-packages/freeopcuaclient/uaclient.py", line 2, in <module>
    from opcua import ua
  File "/usr/local/lib/python3.4/dist-packages/opcua/__init__.py", line 19, in <module>
    from opcua.client import Client
  File "/usr/local/lib/python3.4/dist-packages/opcua/client.py", line 13, in <module>
    from opcua import uacrypto
  File "/usr/local/lib/python3.4/dist-packages/opcua/uacrypto.py", line 1, in <module>
    from Crypto.Util.asn1 import DerSequence
ImportError: No module named 'Crypto'
ml@y13:~$ 
```

On Debian, installing:
```Shell:
$ aptitude install python3-crypto
```
fixed this.

There should be a package dependency, or at least a mention in the Readme, about this.

oroulet, 2015-12-10:
Already fixed

oroulet, 2015-12-10:
Was a bugh in freeopcua

closed 2015-12-10 by oroulet