FreeOpcUa/opcua-client-gui #3: TypeError: QMainWindow.restoreState(QByteArray, int version=0): argument 1 has unexpected type 'unicode'

koppi, 2016-05-01:
On Ubuntu 16.04 I ran:
```bash
~❯ freeopcua-client
Traceback (most recent call last):
  File "/usr/local/bin/freeopcua-client", line 9, in <module>
    load_entry_point('freeopcua-client==0.4.5', 'console_scripts', 'freeopcua-client')()
  File "/usr/local/lib/python2.7/dist-packages/freeopcuaclient/mainwindow.py", line 462, in main
    client = Window()
  File "/usr/local/lib/python2.7/dist-packages/freeopcuaclient/mainwindow.py", line 326, in __init__
    self.restoreState(self.settings.value("main_window_state", b""))
TypeError: QMainWindow.restoreState(QByteArray, int version=0): argument 1 has unexpected type 'unicode'
```
Any ideas, what could cause this?

oroulet, 2016-05-04:
This is the settings stuff. Will have a look

fville, 2016-06-22:
I resolved the error with this change:
`self.restoreState(self.settings.value("main_window_state", b"", type='QByteArray'))`

koppi, 2016-06-22:
Thank's @fville . With the above change I now can open the gui. @oroulet please merge the change and close this issue.

oroulet, 2016-06-23:
@fville thanks but please report issues and even better create pull request!

oroulet, 2016-06-23:
Sorry I had forgotten that bug. Feel free to ping me. Github is not very good at reminding bugs

fville, 2016-06-23:
I created a pull request  (too late apparently): #5 
This is a blind fix because I can't run the opcua-client-gui
I get this:
`Traceback (most recent call last):
  File "freeopcuaclient/mainwindow.py", line 12, in <module>
    from opcua.common.ua_utils import string_to_variant, variant_to_string, val_to_string
ImportError: No module named ua_utils
`

oroulet, 2016-06-23:
sorry for that. the GUI is jsut now depending on a branch og python-opcua which I could not merge because of dependency to datetutil. Will try to fix this asap

oroulet, 2016-06-26:
I just merged the PR in python-opcua, if now use python-opcua from master you will be able to run opcua-client-gui from master too. I fixed many things lately so information is now much better presented for humans and you can write values

koppi, 2016-06-26:
Thank's for opcua-client-gui @oroulet . Closing this issue as resolved.

closed 2016-06-26 by koppi