FreeOpcUa/opcua-client-gui #16: ImportError: No module named uaclient

mitjafelicijan, 2017-02-15:
I have installed gui client with sudo pip install opcua-client and I am getting this error. I have also installed freeopcua with pip.

Is there any way to get this working?

oroulet, 2017-02-18:
uaclient is part of opcu-client and freeopcua should be downloaded automatically, but I see that you are using python 2.7, I am really not sure the GUI works with python 2.7, did pip let you install it?

johndzilva, 2017-04-12:
you should probably try installing the opcua-client using pip3 command install.... I think when you install with pip command, the uaclient.py import will not work. 
And make sure you have latest PyQt installed.

dvucko, 2018-05-08:
I just want to confirm that installing with `pip3` resolves the original issue, at least on my system.

closed 2018-05-08 by oroulet

WangAllen, 2018-12-24:
[image]
what‘s wrong with ’opcua-client‘ transformed to python2？

oroulet, 2018-12-24:
Something broken on you PC. But anyway this is supposed to be s python 3 application

PorkPy, 2019-02-25:
Sorry, I'm late to the party but could this error be because you already have a python2 version installed?
You may need to 'pip uninstall opcua-client' first.
I'm also trying to install opcua-client but my error is:
 
[image]
I suspect I'm missing some dependancy??

zerox1212, 2019-02-25:
Are you trying to run this on Windows?

PorkPy, 2019-02-26:
No, Ubuntu 16.04. I don’t like the burgundy bash.

Shide, 2021-06-24:
You should install `opcua` and `opcua-client` with default python command:

    python -m pip install --upgrade opcua

    python -m pip install --upgrade opcua-client