FreeOpcUa/opcua-client-gui #17: How to method call

aquari89, 2017-04-19:
Hello, All.

I use the freeopcua Gui Client tool for the first time.
How do you call a method in this tool?

I do not know if there is no reference to the method call.

oroulet, 2017-04-21:
no implemented.

oroulet, 2018-05-21:
Implemented in master now

closed 2018-05-21 by oroulet

fabian727, 2018-06-20:
Hi,

nice tool, works good but no "call method". Ive "installed"  the client again. It downloads the 0.7.2 official release version. Is it there already implemented? If so, where is the button (maybe I am too dump to see it)? If not, at which time will you release this version? If it is not tomorrow ;), how can I install your tool from this repo?

Thanks
Fabian

oroulet, 2018-06-20:
Good point. Will release it ASAP

oroulet, 2018-06-20:
Released need version now. Pip install opcua-client --upgrade

oroulet, 2018-06-20:
Let me know if it work y

fabian727, 2018-06-20:
thanks, if business world was as fast as you....
works like a charm.

I have methods with no return values at the moment. If I call them I get the response: 'NoneType' object is not iterable. Would be nice to get a feedback, like "method returned nothing, but calling worked". Your way I was at first glance not sure if it worked or it didn't, but it's a little feature-wish, no must have.

oroulet, 2018-06-20:
Thanks for the feedback I will check that case

zerox1212, 2018-06-20:
If you have control of the server it might be a good idea to have your methods at least return `success = 1`.

oroulet, 2018-06-20:
@fabian727 can you open a new bug request and explain the issue again. put a screenshot of the method node and its children with their values. I did a few tests here and I did not get any errors

fabian727, 2018-06-21:
those aren't errors. I think (didn't read your code) that you save the return values in an array and print all values and types, but if there is no feedback, I get this message

'NoneType' object is not iterable

So the call itself was successfull, but it is not stated because I do not return values and your gui only states there is no feedback
(@zerox1212 I do not want to return values, as it would be minimum a byte again, which crowds the network and yes this byte is important to me)

[image]

oroulet, 2018-06-21:
That message is not normal. This is an error (but it may be a user error) what server do you use?

oroulet, 2018-06-21:
I have had the same error earlier and this was user feil