FreeOpcUa/opcua-client-gui #19: Issue that duplicates objects

aquari89, 2017-04-21:
I am currently using GUI client for testing purposes.

During testing with the GUI client, I found one problem.
When testing with the example server of Unified Automation, there is a problem that objects are duplicated.
This should not happen when testing with PythonOpcUa Server.

I attached the picture of the problem here.
[image]

From a functional point of view I have verified many things. By the way, how do you call method in GUI client?
Has it not been implemented yet?

oroulet, 2017-04-21:
method call is not implemented. 
What do you do to make 2 instances of same node visible? Can you write a scenario?

aquari89, 2017-04-21:
The above server is not my development.
Currently I am using the .Net Sample Server provided by Unified Automation.

(Reference : http://documentation.unified-automation.com/uasdkdotnet/2.5.4/html/L3ServerTutGSLess06.html)

oroulet, 2017-04-24:
fixed in https://github.com/FreeOpcUa/opcua-widgets/commit/84609442f6dcd75ef320c2049781becaf77a4597
not released yet

aquari89, 2017-04-25:
I applied the code you modified and checked freeopcua gui client tool.
It runs well.
[image]

closed 2018-05-21 by oroulet