FreeOpcUa/opcua-client-gui #20: Segmentation fault

freetom, 2017-06-03:
I followed the instruction to install the software. Installed the dependencies like PyQt5. When I try to run it with `opcua-client` I just get a "segmentation fault".

System is Debian 8, testing. Attached valgrind output

[valgrind_log.txt](https://github.com/FreeOpcUa/opcua-client-gui/files/1049489/valgrind_log.txt)

oroulet, 2017-06-03:
Hi, opcua-client is python application. A segfault is nearly never a python
programming error.  There is something broken in your system. Does any
other qt5 application work? Any pyqt5 application?

freetom, 2017-06-03:
I just tested qupzilla, from [here](https://wiki.manjaro.org/index.php?title=List_of_Qt_Applications). It uses Qt5 and seems to work..

MattsSe, 2017-06-19:
It seems that the newest version of pyqt5 (5.8.2) is incompatible.
I downgraded to 5.8.1.1 and everything seems to work again 
`pip install -Iv pyqt5==5.8.1.1` fixed it for me

oroulet, 2017-06-19:
what issue do you get? the API should be stable so it sounds like a qt bug

MattsSe, 2017-06-19:
Just `[1]    1871 segmentation fault  opcua-client` on the console

oroulet, 2017-06-19:
ok this is probably somehting on the distrib. library version osv..

freetom, 2017-06-20:
Using Python2 was probably my issue. With Python3 it works fine

closed 2017-06-20 by freetom