FreeOpcUa/opcua-client-gui #21: Can't browse server's objects

freetom, 2017-06-23:
Hi, 

I am connecting the present OPC UA client to the official [ANSI C server](https://github.com/OPCFoundation/UA-AnsiC). Once connected, I try to browse the folders or do anything else but get errors about a non-activated session. The attached log was the comprehensive output of the client. 

[log.txt](https://github.com/FreeOpcUa/opcua-client-gui/files/1097241/log.txt)

Maybe the server requires more configuration.. 

Thanks in advance

oroulet, 2017-06-23:
Sounds like authentication issue. You need to find out what your server
want. Maybe it requires cryptography...

freetom, 2017-07-30:
.. in the end it was because of the expiration time for the session of ~1 second. If I did not click instantly to interact with the server I get kicked out..

closed 2017-07-30 by freetom