FreeOpcUa/opcua-client-gui #24: Cannot write to 'Byte' attributes

crenwick, 2017-07-31:
When using the GUI to write to a Byte attribute (like changing a 1 to a 2), this error is raised:

```py
struct.error: required argument is not an integer
```

Here is a the logs from an error when changing a 1 to a 2:

```py
uawidgets.attrs_widget - INFO - Writing attribute AttributeIds.Value of node Node(StringNodeId(ns=6;s=::AsGlobalPV:Server.pot[0].setPotState.potState)) with value: DataValue(Value:Variant(val:b'2',type:VariantType.Byte), StatusCode:StatusCode(Good))')
uawidgets.attrs_widget - ERROR - Exception while writing DataValue(Value:Variant(val:b'2',type:VariantType.Byte), StatusCode:StatusCode(Good)) to AttributeIds.Value')
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/site-packages/uawidgets/attrs_widget.py", line 397, in _write_attr
    self.attrs_widget.current_node.set_attribute(data.attr, dv)
  File "/usr/local/lib/python3.6/site-packages/opcua/common/node.py", line 231, in set_attribute
    result = self.server.write(params)
  File "/usr/local/lib/python3.6/site-packages/opcua/client/ua_client.py", line 313, in write
    data = self._uasocket.send_request(request)
  File "/usr/local/lib/python3.6/site-packages/opcua/client/ua_client.py", line 74, in send_request
    future = self._send_request(request, callback, timeout, message_type)
  File "/usr/local/lib/python3.6/site-packages/opcua/client/ua_client.py", line 53, in _send_request
    binreq = request.to_binary()
  File "/usr/local/lib/python3.6/site-packages/opcua/ua/uaprotocol_auto.py", line 9887, in to_binary
    packet.append(self.Parameters.to_binary())
  File "/usr/local/lib/python3.6/site-packages/opcua/ua/uaprotocol_auto.py", line 9836, in to_binary
    packet.append(fieldname.to_binary())
  File "/usr/local/lib/python3.6/site-packages/opcua/ua/uaprotocol_auto.py", line 9792, in to_binary
    packet.append(self.Value.to_binary())
  File "/usr/local/lib/python3.6/site-packages/opcua/ua/uatypes.py", line 988, in to_binary
    packet.append(self.Value.to_binary())
  File "/usr/local/lib/python3.6/site-packages/opcua/ua/uatypes.py", line 835, in to_binary
    b.append(uabin.pack_uatype(self.VariantType, self.Value))
  File "/usr/local/lib/python3.6/site-packages/opcua/ua/ua_binary.py", line 259, in pack_uatype
    return getattr(Primitives, vtype.name).pack(value)
  File "/usr/local/lib/python3.6/site-packages/opcua/ua/ua_binary.py", line 202, in pack
    return struct.pack(self.format, data)
struct.error: required argument is not an integer
uaclient.mainwindow - WARNING - showing error: %s')
```

Though its coming from [this opcua-widgets function](https://github.com/FreeOpcUa/opcua-widgets/blob/master/uawidgets/attrs_widget.py#L394-L403), I wasn't sure what repo to file this issue under.

oroulet, 2017-08-01:
just tested now and writting a byte seems to work. And yes it expects an int between 0 and 254. what else did you try to write? looks like you tried to write something like b"2"
A possible improvment is to support hex notation but I am not sure there is really a need for it. but patch is welcome

crenwick, 2017-08-01:
Here's a screenshot of the attributes:
[image]

Changing that value to anything (0-254, or otherwise), spits out the above error. I think there is something in the GUI or widget that is converting my `2` to a `b'2'` (python's byte format).

Note: I have tested this with another client on a windows machine and I am able to change these values. So I know it's not a OPCUA or server issue.

hayyanon, 2018-05-14:
Hallo , i am using opcua GUI client.
is there any way to customize this type of attributed representation, i need to send/subscribe a query to get only specific node , not all the node on the server side . So to do that i need to customize the code anyhow to be able to tell the server that i need this node (e.g: NodeId = 82), then the query should returns the attributes of the NodeId = 82

thanks in Advance for your Help