FreeOpcUa/opcua-client-gui #27: Black theme

iamjenmackle, 2018-02-01:
Downloaded the client (Mac OSX) and it doesn't look anything like on the screenshot provided here. 
Is it possible to somehow activate that black theme? 

Otherwise great client, works perfectly :)

oroulet, 2018-02-01:
very good question... The application uses Qt5, it is supposed to follow the theme of the desktop it is installed on. It used to work on linux, but now it is white again...
Qt4 had an applicaiton called qtconfig to change color of qt application. It does not exist anymore for qt5....

oroulet, 2018-06-02:
On linux one can do the following
1. Install qt5-styleplugins and qt5ct
2. export QT_QPA_PLATFORMTHEME=qt5ct  # in terminal or .profile or /etc/profile or whatever
3. run qt5ct and select a theme which is black

cotwitch, 2020-04-16:
... and on Mac, macOS respectively?