FreeOpcUa/opcua-client-gui #28: Reset graphic windows

davidemess, 2018-03-12:
Hi, 
first of all, great job! I have a issue with the graphic window, I wrongly close thewindows on the right: Attributes, Events, References, Subscriptions. How can i back to the default settings? I tryed to delete and download again the .zip file, but doesn't work. Thank you in advance.

oroulet, 2018-03-12:
Try to right click many places. Should be possible to get a menu to show it back (standard qt)

oroulet, 2018-03-12:
Otherwise remove/edit the config file which is somewhere under ./config/freeopcua

davidemess, 2018-03-12:
I right clicked under the endpoint and as you have said there is a menu for that. Thank you!

closed 2018-05-21 by oroulet

saizen408, 2020-04-28:
there's no obvious UI element to allow for resetting the windows. I had to edit the installation source code that controls this auto-save functionality by going to /Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/uaclient/mainwindow.py and adding self.settings.clear() on line 238 to clear the auto-saved window settings right after the QSettings() class is instantiated. Load up the client again and all the windows should be reset. Don't forget to remove self.settings.clear() afterwards.