FreeOpcUa/opcua-client-gui #29: No module named 'uawidgets'

WangAllen, 2018-04-16:
SYS:PATH ['C:\\Users\\39544\\Desktop\\opcua-client-gui-master', 'D:\\Program Files\\Python36\\python36.zip', 'D:\\Program Files\\Python36\\DLLs', 'D:\\Program Files\\Python36\\lib', 'D:\\Program Files\\Python36', 'C:\\Users\\39544\\AppData\\Roaming\\Python\\Python36\\site-packages', 'D:\\Program Files\\Python36\\lib\\site-packages']
PWD C:\Users\39544\Desktop\opcua-client-gui-master
cryptography is not installed, use of crypto disabled
cryptography is not installed, use of crypto disabled
Traceback (most recent call last):
  File "C:\Users\39544\Desktop\opcua-client-gui-master\tests.py", line 17, in <module>
    from uaclient.mainwindow import Window
  File "C:\Users\39544\Desktop\opcua-client-gui-master\uaclient\mainwindow.py", line 19, in <module>
    from uaclient.connection_dialog import ConnectionDialog
  File "C:\Users\39544\Desktop\opcua-client-gui-master\uaclient\connection_dialog.py", line 4, in <module>
    from uawidgets.utils import trycatchslot
ModuleNotFoundError: No module named 'uawidgets'

oroulet, 2018-04-16:
You need to install dependencies. Look into setup.py. maybe you could also update readme to list dependencies when running from source

WangAllen, 2018-04-16:
Thank you a lot, I didn't download opcua-widgets fisrt.but now there is another question,how could i use uaclient as package.if using "from uaclient.uaclient import UaClient" was wrong, while using "from .uaclient import UaClient" maybe right.

cryptography is not installed, use of crypto disabled
cryptography is not installed, use of crypto disabled
Traceback (most recent call last):
  File "C:\Users\39544\Desktop\opcua-client-gui-master\uaclient\mainwindow.py", line 17, in <module>
    from uaclient.uaclient import UaClient
ModuleNotFoundError: No module named 'uaclient.uaclient'; 'uaclient' is not a package

and how could i install the cryptography?

WangAllen, 2018-04-16:
E:\Program Files (x86)\WinPython> python -m pip install .\opcua-client-gui-master\
Processing e:\program files (x86)\winpython\opcua-client-gui-master
Requirement already satisfied: opcua>=0.95.1 in e:\program files (x86)\winpython\python-3.6.5.amd64\lib\site-packages (from opcua-client==0.7.2) (0.95.1)
Requirement already satisfied: opcua-widgets>=0.5.0 in e:\program files (x86)\winpython\python-3.6.5.amd64\lib\site-packages (from opcua-client==0.7.2) (0.5.2)
Requirement already satisfied: lxml in e:\program files (x86)\winpython\python-3.6.5.amd64\lib\site-packages (from opcua>=0.95.1->opcua-client==0.7.2) (4.2.1)
Requirement already satisfied: pytz in e:\program files (x86)\winpython\python-3.6.5.amd64\lib\site-packages (from opcua>=0.95.1->opcua-client==0.7.2) (2018.3)
Requirement already satisfied: python-dateutil in e:\program files (x86)\winpython\python-3.6.5.amd64\lib\site-packages (from opcua>=0.95.1->opcua-client==0.7.2) (2.7.2)
Requirement already satisfied: six>=1.5 in e:\program files (x86)\winpython\python-3.6.5.amd64\lib\site-packages (from python-dateutil->opcua>=0.95.1->opcua-client==0.7.2) (1.11.0)
Building wheels for collected packages: opcua-client
  Running setup.py bdist_wheel for opcua-client ... done
  Stored in directory: C:\Users\39544\AppData\Local\pip\Cache\wheels\d8\8d\3a\4a3c76d76f29ed8f37758af1e847edf76068a6dfa5f615c41d
Successfully built opcua-client
tensorflow-tensorboard 1.5.1 has requirement bleach==1.5.0, but you'll have bleach 2.1.3 which is incompatible.
tensorflow-tensorboard 1.5.1 has requirement html5lib==0.9999999, but you'll have html5lib 1.0.1 which is incompatible.
Installing collected packages: opcua-client
  Found existing installation: opcua-client 0.7.2
    Uninstalling opcua-client-0.7.2:
      Successfully uninstalled opcua-client-0.7.2
Successfully installed opcua-client-0.7.2

is this meaning i had created successsfully?but i didn't find the opcua-client.exe

oroulet, 2018-04-16:
looks likeit was successfull. not sure where things  are installed under windows. look under Scripts folder in your pythoin installation maybe

WangAllen, 2018-04-16:
Thank you at all,i failed to find it in windows.
I now use ubantu,I can run it with "How To Install",if i have a local opc server,how could i make them connect,by changing the code with "URL" to my local server address?

closed 2018-05-21 by oroulet

BhargavN02, 2022-12-03:
ModuleNotFoundError: No module named 'uawidgets', this error keep popping while running the app.py file