FreeOpcUa/opcua-client-gui #31: AttributeError: 'NoneType' object has no attribute 'to_string'

wiewioraseb, 2018-05-21:
Hi,
I wanted to test opcua-client-gui with OPC Fundation example `ServerExample1` in repo: [ServerExample1.java](https://github.com/OPCFoundation/UA-Java/blob/master/examples/basic/src/main/java/org/opcfoundation/ua/examples/ServerExample1.java)

While FreeOpcUa Client and ServerExample1 is running and I press 'Connect' on the client UI, I get error 

> AttributeError: 'NoneType' object has no attribute 'to_string'

Whole log from client: 

> uaclient.uaclient - INFO - Connecting to opc.tcp://localhost:8666/UAExample with parameters None, None, None, None')
> uawidgets.utils - ERROR - 'NoneType' object has no attribute 'to_string'')
> Traceback (most recent call last):
>   File "d:\python35\lib\site-packages\uawidgets\utils.py", line 21, in wrapper
>     result = func(self, *args)
>   File "d:\python35\lib\site-packages\uaclient\mainwindow.py", line 338, in connect
>     self.tree_ui.set_root_node(self.uaclient.client.get_root_node())
>   File "d:\python35\lib\site-packages\uawidgets\tree_widget.py", line 43, in set_root_node
>     self.model.set_root_node(node)
>   File "d:\python35\lib\site-packages\uawidgets\tree_widget.py", line 164, in set_root_node
>     self.add_item(desc, node=node)
>   File "d:\python35\lib\site-packages\uawidgets\tree_widget.py", line 177, in add_item
>     item = [QStandardItem(desc.DisplayName.to_string()), QStandardItem(desc.BrowseName.to_string()), QStandardItem(desc.NodeId.to_string())]
> AttributeError: 'NoneType' object has no attribute 'to_string'
> uaclient.mainwindow - WARNING - showing error: %s')

Am I missing some connection options? Or some dependencies?

Thank you in advance for your help.

oroulet, 2018-05-21:
Looks like the root node of your server is either missing display name or browse name. Looks like a broken server. But leave bug open. We should not crash on such error

oroulet, 2018-05-29:
theoretical fix here: https://github.com/FreeOpcUa/opcua-widgets/commit/70700b8ec0eedbe9323718d8cd5047b5e7ef55c4 It is not released yet

closed 2018-06-20 by oroulet