FreeOpcUa/opcua-client-gui #32: On Mac, cannot reopen Attribute/Graph etc widgets after close it

yangwang166, 2018-06-18:
On Mac, once I close the Attribute/Graph/Event widgets, I cannot find how to reopen it.
I have tried to reinstalled the opcua-client, but didn't work.

oroulet, 2018-06-18:
right click (Ctrl click on mac) everywhere. There is a menu somewhere. Otherwise you need to delete configuration file which is a the standard Mac place  (But I have no idea where)

yangwang166, 2018-06-18:
With Rgiht click, I saw the menu, but the item in the menu only have:
* Subscribe to data change
* UnSubscribe to data change
* Subscribe to Event
* UnSubscribe to Event
* Add to Graph
* Remove from Graph
* Copy Path
* Copy NodeId.

Cannot find item related to reopen the widget.
I think I need to find out where the configuration file is.

oroulet, 2018-06-18:
Do not right clock on the tree. Right click everywhere on interfaces: toolbars, split, etc...

yangwang166, 2018-06-18:
There is no menu when I right click on toolbars or split.

oroulet, 2018-06-18:
[image]

at least on linux there is that extra Qt menu.

oroulet, 2018-06-18:
[image]

At least on linux there is a menu on toolbar

yangwang166, 2018-06-18:
Thank you, i think it maybe an incompatible issue in mac.

yangwang166, 2018-06-20:
I couldn't find the configuration file for qt in mac.
But I finally commented the following code for load the setting, then the dashboard back to work.

```python
275         Don't memory setting
276         data = self.settings.value("main_window_state", None)
277         if data:
278             self.restoreState(data)
```

closed 2018-06-20 by oroulet