FreeOpcUa/opcua-client-gui #36: Install GUI on Rasperry Pi

pablofsmelo, 2019-02-13:
Hello,

I'm trying to install opcua-client-gui on my Raspberry Pi, but I'm having problems. After that I installed PyQT5, I started installing the GUI and I got the following error:

Collecting opcua-client
  Downloading https://files.pythonhosted.org/packages/de/99/41aa0eb2b9ff2560e6156e8d0a5ce294f5d2aa7d10232dd23485bbe099ac/opcua-client-0.8.0.tar.gz (204kB)
    100% |████████████████████████████████| 215kB 202kB/s 
Collecting opcua-widgets>=0.5.4 (from opcua-client)
  Downloading https://files.pythonhosted.org/packages/5a/e3/1fe068af0e0ad3edd4baed6aadec04099a0384df5d78fd13b3490f520f63/opcua-widgets-0.5.6.tar.gz (209kB)
    100% |████████████████████████████████| 215kB 190kB/s 
Collecting opcua>=0.98.1 (from opcua-client)
  Downloading https://files.pythonhosted.org/packages/79/fc/dca3c2cc6760562e7c81995ab55ab0d0300db604eb839f549aabd9d816d1/opcua-0.98.6.tar.gz (450kB)
    100% |████████████████████████████████| 460kB 151kB/s 
Requirement already satisfied: enum34 in /usr/lib/python2.7/dist-packages (from opcua>=0.98.1->opcua-client)
Collecting futures (from opcua>=0.98.1->opcua-client)
  Downloading https://files.pythonhosted.org/packages/2d/99/b2c4e9d5a30f6471e410a146232b4118e697fa3ffc06d6a65efde84debd0/futures-3.2.0-py2-none-any.whl
Collecting lxml (from opcua>=0.98.1->opcua-client)
  Downloading https://files.pythonhosted.org/packages/0f/bd/bb1464d1f363dbd805862c8a1ea258b9a4f4d2049c376d1c4790b6545691/lxml-4.3.1.tar.gz (4.4MB)
    100% |████████████████████████████████| 4.4MB 59kB/s 
Collecting python-dateutil (from opcua>=0.98.1->opcua-client)
  Using cached https://files.pythonhosted.org/packages/41/17/c62faccbfbd163c7f57f3844689e3a78bae1f403648a6afb1d0866d87fbb/python_dateutil-2.8.0-py2.py3-none-any.whl
Collecting pytz (from opcua>=0.98.1->opcua-client)
  Downloading https://files.pythonhosted.org/packages/61/28/1d3920e4d1d50b19bc5d24398a7cd85cc7b9a75a490570d5a30c57622d34/pytz-2018.9-py2.py3-none-any.whl (510kB)
    100% |████████████████████████████████| 512kB 179kB/s 
Collecting trollius (from opcua>=0.98.1->opcua-client)
  Downloading https://files.pythonhosted.org/packages/98/47/000d403a209e5d0c2753feabd3bbbd09a11e32652899abffc1ef5d5c0abb/trollius-2.2.tar.gz (276kB)
    100% |████████████████████████████████| 276kB 200kB/s 
Requirement already satisfied: six>=1.5 in /usr/lib/python2.7/dist-packages (from python-dateutil->opcua>=0.98.1->opcua-client)
Building wheels for collected packages: opcua-client, opcua-widgets, opcua, lxml, trollius
  Running setup.py bdist_wheel for opcua-client ... done
  Stored in directory: /root/.cache/pip/wheels/ff/de/69/3870227c085b3ebd1fa71a623dca95be418780c547b47081b4
  Running setup.py bdist_wheel for opcua-widgets ... done
  Stored in directory: /root/.cache/pip/wheels/1e/1c/86/5867f1b95812bca3d57b37b13845798f75f468376f6d057266
  Running setup.py bdist_wheel for opcua ... done
  Stored in directory: /root/.cache/pip/wheels/89/1e/0e/ca3ed5e56a4b6d91928133c62956767da8c56dac9873dcdfe3
  Running setup.py bdist_wheel for lxml ... error
  Complete output from command /usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-uYeHCA/lxml/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/tmpGgfN5Bpip-wheel- --python-tag cp27:
  Building lxml version 4.3.1.
  Building without Cython.
  ERROR: /bin/sh: 1: xslt-config: not found
  
  ** make sure the development packages of libxml2 and libxslt are installed **
  
  Using build configuration of libxslt
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.linux-armv7l-2.7
  creating build/lib.linux-armv7l-2.7/lxml
  copying src/lxml/doctestcompare.py -> build/lib.linux-armv7l-2.7/lxml
  copying src/lxml/_elementpath.py -> build/lib.linux-armv7l-2.7/lxml
  copying src/lxml/cssselect.py -> build/lib.linux-armv7l-2.7/lxml
  copying src/lxml/__init__.py -> build/lib.linux-armv7l-2.7/lxml
  copying src/lxml/builder.py -> build/lib.linux-armv7l-2.7/lxml
  copying src/lxml/usedoctest.py -> build/lib.linux-armv7l-2.7/lxml
  copying src/lxml/pyclasslookup.py -> build/lib.linux-armv7l-2.7/lxml
  copying src/lxml/sax.py -> build/lib.linux-armv7l-2.7/lxml
  copying src/lxml/ElementInclude.py -> build/lib.linux-armv7l-2.7/lxml
  creating build/lib.linux-armv7l-2.7/lxml/includes
  copying src/lxml/includes/__init__.py -> build/lib.linux-armv7l-2.7/lxml/includes
  creating build/lib.linux-armv7l-2.7/lxml/html
  copying src/lxml/html/diff.py -> build/lib.linux-armv7l-2.7/lxml/html
  copying src/lxml/html/__init__.py -> build/lib.linux-armv7l-2.7/lxml/html
  copying src/lxml/html/soupparser.py -> build/lib.linux-armv7l-2.7/lxml/html
  copying src/lxml/html/formfill.py -> build/lib.linux-armv7l-2.7/lxml/html
  copying src/lxml/html/_html5builder.py -> build/lib.linux-armv7l-2.7/lxml/html
  copying src/lxml/html/_setmixin.py -> build/lib.linux-armv7l-2.7/lxml/html
  copying src/lxml/html/_diffcommand.py -> build/lib.linux-armv7l-2.7/lxml/html
  copying src/lxml/html/builder.py -> build/lib.linux-armv7l-2.7/lxml/html
[... 151 lines cut ...]
    building 'lxml.etree' extension
    creating build/temp.linux-armv7l-2.7
    creating build/temp.linux-armv7l-2.7/src
    creating build/temp.linux-armv7l-2.7/src/lxml
    arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -Wdate-time -D_FORTIFY_SOURCE=2 -g -fdebug-prefix-map=/build/python2.7-Ub3vap/python2.7-2.7.13=. -fstack-protector-strong -Wformat -Werror=format-security -fPIC -DCYTHON_CLINE_IN_TRACEBACK=0 -Isrc -Isrc/lxml/includes -I/usr/include/python2.7 -c src/lxml/etree.c -o build/temp.linux-armv7l-2.7/src/lxml/etree.o -w
    In file included from src/lxml/etree.c:687:0:
    src/lxml/includes/etree_defs.h:14:31: fatal error: libxml/xmlversion.h: No such file or directory
     #include "libxml/xmlversion.h"
                                   ^
    compilation terminated.
    Compile failed: command 'arm-linux-gnueabihf-gcc' failed with exit status 1
    cc -I/usr/include/libxml2 -c /tmp/xmlXPathInittu2CgT.c -o tmp/xmlXPathInittu2CgT.o
    /tmp/xmlXPathInittu2CgT.c:1:26: fatal error: libxml/xpath.h: No such file or directory
     #include "libxml/xpath.h"
                              ^
    compilation terminated.
    *********************************************************************************
    Could not find function xmlCheckVersion in library libxml2. Is libxml2 installed?
    *********************************************************************************
    error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1
    
    ----------------------------------------
Command "/usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-uYeHCA/lxml/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-rnysqr-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-uYeHCA/lxml

Could anyonde tell me what this means ?

oroulet, 2019-02-13:
You have some issues with lxml. Try to find a compiled version of python-lxml. But anyway the GUI is a python 3 only application and you are using python 2

zerox1212, 2019-02-13:
Unfortunately `lxml` has dependencies on native C libraries. You should do some google searches for `lxml` specifically.

pablofsmelo, 2019-02-13:
I just upgraded to Python 3 and it worked. 

[image]

Thank you :)

closed 2019-02-13 by pablofsmelo