FreeOpcUa/opcua-client-gui #37: Error by opcua-client command

akkbng, 2019-03-30:
Hello,

I'm trying to use opcua-client-gui on my Raspberry pi 3. I installed pyqt5 and opcua-client with pip without a problem but when i run the "opcua-client" command in the terminal i get the error below:

> Traceback (most recent call last):
  File "/usr/local/bin/opcua-client", line 7, in <module>
    from uaclient.mainwindow import main
  File "/usr/local/lib/python3.5/dist-packages/uaclient/mainwindow.py", line 23, in <module>
    from uawidgets.attrs_widget import AttrsWidget
  File "/usr/local/lib/python3.5/dist-packages/uawidgets/attrs_widget.py", line 11, in <module>
    from uawidgets.get_node_dialog import GetNodeButton
  File "/usr/local/lib/python3.5/dist-packages/uawidgets/get_node_dialog.py", line 7, in <module>
    from uawidgets.tree_widget import TreeWidget
  File "/usr/local/lib/python3.5/dist-packages/uawidgets/tree_widget.py", line 62
    raise ValueError(f"Node {node} not found in tree")
                                                    
SyntaxError: invalid syntax

I did a quick google search but couldn't understand what the problem is. I would really appreciate your help on this, thanks in advance :)

oroulet, 2019-03-30:
seems to be a code with python3.6 syntax. you need to replace the f"" string with the "".format stuff

zerox1212, 2019-03-31:
I would make sure your Pi has the latest version of python and you are actually running it in Python 3. If I remember right the Pi out of the box comes with a really old version of python.

JacksonJoy, 2019-05-06:
Hi anyone found the solution for this problem. I am also facing the same issue. Coudl anyone please help
pi@raspberrypi:~ $ opcua-client
pyqtgraph or numpy are not installed, use of graph feature disabled
Traceback (most recent call last):
  File "/usr/local/bin/opcua-client", line 7, in <module>
    from uaclient.mainwindow import main
  File "/usr/local/lib/python3.5/dist-packages/uaclient/mainwindow.py", line 23, in <module>
    from uawidgets.attrs_widget import AttrsWidget
  File "/home/pi/.local/lib/python3.5/site-packages/uawidgets/attrs_widget.py", line 11, in <module>
    from uawidgets.get_node_dialog import GetNodeButton
  File "/home/pi/.local/lib/python3.5/site-packages/uawidgets/get_node_dialog.py", line 7, in <module>
    from uawidgets.tree_widget import TreeWidget
  File "/home/pi/.local/lib/python3.5/site-packages/uawidgets/tree_widget.py", line 62
    raise ValueError(f"Node {node} not found in tree")
                                                    ^
SyntaxError: invalid syntax

joesan, 2019-06-28:
I'm facing the same issue! Is there a fix for this?

```
Joes-MacBook-Pro:bin joesan$ python opcua-client
No handlers could be found for logger "opcua.client.client"
Traceback (most recent call last):
  File "opcua-client", line 6, in <module>
    from uaclient.mainwindow import main
  File "/usr/local/lib/python2.7/site-packages/uaclient/mainwindow.py", line 17, in <module>
    from uaclient.uaclient import UaClient
ImportError: No module named uaclient
Joes-MacBook-Pro:bin joesan$
```

joesan, 2019-06-28:
I managed to get this running using pip3:

`pip install opcua-client`

and then:

`python3 opcua-client`

oroulet, 2019-06-28:
That issue is completely different. Read the trace. This is an installation error from user @joesan

shraddha291996, 2020-08-05:
Hello,
While installing OPCUA connector kafka I am getting this error I have attached. I tried searching on google but could not find anything to solve the error. Could anyone please help :) Thanks in advance.
[image]

zerox1212, 2020-08-05:
@shraddha291996 Make a new issue. This issue has nothing to do with kafka. Also don't double post your problem in multiple issues. This is just going to annoy the people that might help you and the owner of the repo is should probably delete your question.

closed 2022-05-26 by akkbng