FreeOpcUa/opcua-client-gui #38: Error by OPCUA-Client command after installing all packages

JacksonJoy, 2019-05-06:
Hi,
I am trying to make raspberry pi 3 as client to visualize my data. 
At the final step after installing opcua and opcua-client **i cannot run the opc-ua client.** 

It is showing invalid syntax. I am using Python 3.5.3. Is the syntax different. I am stuck. Could someone please help

pi@raspberrypi:~ $ opcua-client
pyqtgraph or numpy are not installed, use of graph feature disabled
Traceback (most recent call last):
  File "/usr/local/bin/opcua-client", line 7, in <module>
    from uaclient.mainwindow import main
  File "/usr/local/lib/python3.5/dist-packages/uaclient/mainwindow.py", line 23, in <module>
    from uawidgets.attrs_widget import AttrsWidget
  File "/home/pi/.local/lib/python3.5/site-packages/uawidgets/attrs_widget.py", line 11, in <module>
    from uawidgets.get_node_dialog import GetNodeButton
  File "/home/pi/.local/lib/python3.5/site-packages/uawidgets/get_node_dialog.py", line 7, in <module>
    from uawidgets.tree_widget import TreeWidget
  File "/home/pi/.local/lib/python3.5/site-packages/uawidgets/tree_widget.py", line 62
    raise ValueError(f"Node {node} not found in tree")
                                                    ^
SyntaxError: invalid syntax

zerox1212, 2019-05-06:
Do you have all dependencies such s PyQT 5?

oroulet, 2019-05-06:
This is a python 3.6 that went in....  F-string is 3.6

JacksonJoy, 2019-05-06:
> Do you have all dependencies such s PyQT 5?

Thank you zerox1212. Yes I have installed PyQT5. It went succesfully.

JacksonJoy, 2019-05-06:
> This is a python 3.6 that went in.... F-string is 3.6

Thank you oroulet. But my version is Python 3.5.3. Should I update my python in raspberry pi to 3.6?

zerox1212, 2019-05-06:
Yes, you need python 3.6 or higher.

cbergmiller, 2019-05-07:
From the stack trace it looks like the f-string is in _uawidgets_
In the master branch there is no f-string in tree_widgets.py:
https://github.com/FreeOpcUa/opcua-widgets/blob/master/uawidgets/tree_widget.py
How can that be?

oroulet, 2019-05-07:
He uses the released version and I probably fixed master

JacksonJoy, 2019-05-07:
Hi. I edited the tree_widget.py, so that it can run on python 3.5. Then it functioned. Thank you everyone for the help

Knight9206, 2019-05-30:
@JacksonJoy 
hi, i also have some problem about in raspberry pi 3+.
how did you deit your tree_widget.py. I try to upgrade python 3.6 and i can not resolve that problem.
can you give me some comment?

vivaladav, 2020-06-18:
Replacing all the occurrences of **(f"** with (" in _uawidgets/tree_widget.py_ allows to run it fine with Python 3.5.

For example  
_raise ValueError(f"Node {node} not found in tree")_
should become
_raise ValueError("Node {node} not found in tree")_

@Knight9206

shraddha291996, 2020-08-05:
Hello,
While installing OPCUA connector kafka I am getting this error I have attached. I tried searching on google but could not find anything to solve the error. Could anyone please help :) Thanks in advance.
[image]

rupesharora, 2023-07-20:
I encountered with same issue. Resolved by running following commands.
`sudo apt install pyqt5-dev-tools pyqt5-dev`

If still doesn't work try below too:
`sudo apt-get install python3-pyqt5`