FreeOpcUa/opcua-client-gui #41: Browsepath

aschaepper, 2019-08-23:
Hello together

I just used the export browsepath functionality which is quite handy. So I have a question reagarding it.

Is there a reason you use , "commas" instead of / "slashes". As far as I can see commas are not according to the standard and I have to replace them with the /
Here is a link:
http://documentation.unified-automation.com/uasdkdotnet/2.0.0/L3ClientTutExample13.html

Also, there should be a / in the beginning of the browse path for signaling the root of the path, otherwise it would be a relative path.

so this path from the opcua-client:
0:Root,0:Objects,2:DeviceSet,2:X20BC008U,2:ParameterSet,2:ConfigurationStatus

Should look like this:
/0:Root/0:Objects/2:DeviceSet/2:X20BC008U/2:ParameterSet/2:ConfigurationStatus

But I could be wrong, thanks anyway for the great software!

oroulet, 2019-08-23:
I though I wrote that because you can past it directly into pyton code as a string mynode.get_child(PASTE HERE). Something I needed quite often. But now I see that it does not work since you need " around each strings.... and you have a point that the standard uses /  .    So we should add support for / in get_child method and split by /. PR welcome ;-)

oroulet, 2019-08-23:
so it must have worked somewhere since I wrote that... but using / seems like a good proposition although we need a backward compatible PR