FreeOpcUa/opcua-client-gui #49: Fails to start

ctron, 2020-07-08:
~~~
$ pip install --user opcua-client
Collecting opcua-client
  Downloading https://files.pythonhosted.org/packages/de/99/41aa0eb2b9ff2560e6156e8d0a5ce294f5d2aa7d10232dd23485bbe099ac/opcua-client-0.8.0.tar.gz (204kB)
     |████████████████████████████████| 215kB 3.1MB/s 
Collecting opcua>=0.98.1
  Downloading https://files.pythonhosted.org/packages/58/fa/c50e971be6377258b3c7917303893dea204c8f958553dc4dccd75015caf0/opcua-0.98.12.tar.gz (571kB)
     |████████████████████████████████| 573kB 8.5MB/s 
Collecting opcua-widgets>=0.5.4
  Downloading https://files.pythonhosted.org/packages/30/e0/7559420565340ff62a962f9e7417a9036edf4b2918b101147f70af496588/opcua-widgets-0.5.9.tar.gz (211kB)
     |████████████████████████████████| 215kB 11.4MB/s 
Requirement already satisfied: python-dateutil in /usr/lib/python3.8/site-packages (from opcua>=0.98.1->opcua-client) (2.8.0)
Requirement already satisfied: pytz in /usr/lib/python3.8/site-packages (from opcua>=0.98.1->opcua-client) (2020.1)
Requirement already satisfied: lxml in /usr/lib64/python3.8/site-packages (from opcua>=0.98.1->opcua-client) (4.4.1)
Requirement already satisfied: six>=1.5 in /usr/lib/python3.8/site-packages (from python-dateutil->opcua>=0.98.1->opcua-client) (1.14.0)
Installing collected packages: opcua, opcua-widgets, opcua-client
    Running setup.py install for opcua ... done
    Running setup.py install for opcua-widgets ... done
    Running setup.py install for opcua-client ... done
Successfully installed opcua-0.98.12 opcua-client-0.8.0 opcua-widgets-0.5.9
[jreimann@brocken ~]$ opcua-client 
Traceback (most recent call last):
  File "/home/jreimann/.local/bin/opcua-client", line 11, in <module>
    load_entry_point('opcua-client==0.8.0', 'console_scripts', 'opcua-client')()
  File "/usr/lib/python3.8/site-packages/pkg_resources/__init__.py", line 489, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python3.8/site-packages/pkg_resources/__init__.py", line 2852, in load_entry_point
    return ep.load()
  File "/usr/lib/python3.8/site-packages/pkg_resources/__init__.py", line 2443, in load
    return self.resolve()
  File "/usr/lib/python3.8/site-packages/pkg_resources/__init__.py", line 2449, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/home/jreimann/.local/lib/python3.8/site-packages/uaclient/mainwindow.py", line 10, in <module>
    from PyQt5.QtCore import pyqtSignal, QTimer, Qt, QObject, QSettings, QItemSelection, QMimeData, QCoreApplication
ModuleNotFoundError: No module named 'PyQt5'

~~~

oroulet, 2020-07-09:
as the error mesage says you need to install pyqt5.python3-pyqt5 if you use some debian based distro

ctron, 2020-07-09:
> as the error mesage says you need to install pyqt5.python3-pyqt5 if you use some debian based distro

Shouldn't the be a "requirement" on the, like for the other dependencies as well?

~~~
Requirement already satisfied: python-dateutil in /usr/lib/python3.8/site-packages (from opcua>=0.98.1->opcua-client) (2.8.0)
Requirement already satisfied: pytz in /usr/lib/python3.8/site-packages (from opcua>=0.98.1->opcua-client) (2020.1)
Requirement already satisfied: lxml in /usr/lib64/python3.8/site-packages (from opcua>=0.98.1->opcua-client) (4.4.1)
Requirement already satisfied: six>=1.5 in /usr/lib/python3.8/site-packages (from python-dateutil->opcua>=0.98.1->opcua-client) (1.14.0)
~~~

oroulet, 2020-07-09:
I was afraid to list it and trigger compilation of pyqt5 for people. and the error is very obvious.... but maybe this is not an issue anymore..

moimart1, 2021-02-19:
I know the error is obvious but can you at least update the doc ?
It's avoid additional research

oroulet, 2021-02-20:
I think it has been added as a requirement now. As far as I remember.