FreeOpcUa/opcua-client-gui #53: What about the client application certificate?

CuriousCrook, 2020-09-09:
A certificate and a private key are specified in the ConnectionDialog. Since these certificate and key are configured per connection, I assume that this is the user certificate, etc.

When trying to establish a connection to an OPC UA server with the following security profile, I am at a loss where the application certificate has to be placed.
SecurityMode: SignAndEncrypt
SecurityPolicy: Basic256Sha256

Usually the application comes with its own certificate. But I couldn't find any in the repository.
With UaExpert, for example, a certificate is generated during installation. There is also a certificate management dialog in which the client application certificate can be renewed. Setting up a connection from UaExpert to this server was no problem.

Could it be that the security profile I am using is not supported by opcua-client-gui?

oroulet, 2020-09-09:
the client does not have its own certificate you need to generate one by hand and specify it in the dialog.
Of course one could make an UI to generate UI and store it somewhere

CuriousCrook, 2021-09-24:
Oh, I assumed that the certificate in the connection dialogue was the user certificate. Afaik the application certificate should not be connection specific. I would have expected the client application certificate somewhere else.

However, I have extended the GUI to manage client application certificates and connection specific user certificates. I just wanted to integrate the change into the current master. However, this really no longer seems to work. I can reproduce some of the issues already reported. I have just commented on these issues as well. 

Are you interested in the extension described above?

With kind regards
CC

CuriousCrook, 2022-01-12:
@oroulet I have made a few extensions to support the security profile suggested by the BSI:
which is:
Security Mode: SignAndEncrypt
Security Policy: Basic256Sha256
User Authentication: Certificate X.509 v3

You can find a running version in my fork:
https://github.com/CuriousCrook/opcua-client-gui/tree/CC-Add-ApplicationCertificate-Support

It is not the nicest implementation in terms of operating system independent PKI directory access. The extension also does not support the automatic generation of client application certificates. 

But it is simple and functional.

Are you interested in a pull request?

With kind regards
CC

oroulet, 2022-01-12:
Can't you make a PR so we can easily see the diff? IT would be a nice feature

CuriousCrook, 2022-01-12:
Sure:
https://github.com/FreeOpcUa/opcua-client-gui/pull/71