FreeOpcUa/opcua-client-gui #55: How to active the opcuaclient in Linux system

jake1026, 2020-11-26:
I have installed the PYQT5 and opcua client, but when I run the command "uaclient"
it say'Traceback (most recent call last):
  File "/usr/local/bin/uaclient", line 11, in <module>
    sys.exit(uaclient())
  File "/home/sinoma-ld/.local/lib/python3.6/site-packages/opcua/tools.py", line 477, in uaclient
    client.connect()
  File "/home/sinoma-ld/.local/lib/python3.6/site-packages/opcua/client/client.py", line 272, in connect
    self.connect_socket()
  File "/home/sinoma-ld/.local/lib/python3.6/site-packages/opcua/client/client.py", line 307, in connect_socket
    self.uaclient.connect_socket(self.server_url.hostname, self.server_url.port)
  File "/home/sinoma-ld/.local/lib/python3.6/site-packages/opcua/client/ua_client.py", line 266, in connect_socket
    return self._uasocket.connect_socket(host, port)
  File "/home/sinoma-ld/.local/lib/python3.6/site-packages/opcua/client/ua_client.py", line 155, in connect_socket
    sock = socket.create_connection((host, port), timeout=self.timeout)
  File "/usr/lib/python3.6/socket.py", line 724, in create_connection
    raise err
  File "/usr/lib/python3.6/socket.py", line 713, in create_connection
    sock.connect(sa)
ConnectionRefusedError: [Errno 111] Connection refused
'

oroulet, 2020-11-26:
well. you do not have any server running or wrong ip addr/port

jake1026, 2020-11-27:
> well. you do not have any server running or wrong ip addr/port

Does it have a GUI screen? I thought it was configured on the open page,
How to use command to active the GUI

Thank very much from China...Jake

JoeyFaulkner, 2020-12-01:
Not sure if this is the problem @jake1026 but on my ubuntu machine i have to run `opcua-client` not `uaclient` :)