FreeOpcUa/opcua-client-gui #59: Custom Stucture as Argument in Method is not shown

AndreyKl4, 2021-01-11:
Hi all!

I have defined a structure as OutputArgument in Method. In UA Expert Client it looks like this:
[image]
In opcua-client-gui it looks like this:
[image]

It seems like opcua-client-gui doesn't  support custom structure as Arguments in Methods... or I done smth wrong?

Thanks in advance!

Andrii Klimov

oroulet, 2021-01-11:
yes. This is probably missing. I never have had to do that.

AndreyKl4, 2021-01-11:
Thanks for the quick answer!

zerox1212, 2021-01-12:
Unless someone did a lot of work, methods in general are very basic for the GUI. Lots of improvements could be done.