FreeOpcUa/opcua-client-gui #61: General Questions

derme302, 2021-02-16:
@oroulet Hey I was hoping you could reach out by email, I just have a couple questions about the package and best way to implement some features. I couldn't find your email, feel free to close this issue.

<derme@outlook.com>

oroulet, 2021-02-17:
Ask the questions here. You will get better and faster answer and help others

derme302, 2021-02-17:
I just had some general images about the package.

1) I am looking to add certificate generation, and was just curious about pulling in other dependencies. In previous issues you've been hesitant to pull in additional libraries. Would adding OpenSSL be okay, or would you rather have the code inline?

2) So with the above and a couple other features that require UI changes, I noticed in other issues that you've minimised adding UI changes. Is there a general direction or intention you have around the UI? Qt Designer is a pain to work with and I don't want to do too much re-work with it.

3) With the last PR/commit to master I was hoping you could do me a favor and amend the commit message to remove the co-author - I made a mistake in my settings.

zerox1212, 2021-02-17:
Do you even need to use Qt Designer? Pretty sure all the GUI is done in code only...

If you don't want to use Qt what do you propose. Personally I do not think there are many good UI packages for Python, which only leaves web based platforms.

derme302, 2021-02-17:
> Do you even need to use Qt Designer? Pretty sure all the GUI is done in code only...

Editing the .ui files is in Qt, I don't mind it, it's just that Qt can just be unintuitive at times.

> If you don't want to use Qt what do you propose? Personally I do not think there are many good UI packages for Python, which only leaves web based platforms.

Yeah, I agree that Qt is pretty much the best that there is. I think I am just curious about the philosophy behind the UI. For instance, adding extra windows (e.g. certificate configuration), ect or if the preference is to keep all the UI to the current (2?) windows.

derme302, 2021-02-20:
Hey, @oroulet I could really use some pointers here. I am keen to make PR for certificate generation but would like to know if I should add it here or against uawidgets?

Also, I could use some help with removing the co-author from that last commit message.

oroulet, 2021-02-20:
Sorry I mised you last messages for some reasons.
* I do not think you need to add any dependency to the gui. The opcua library can handle certificate already. Unless you want to do something special.
* The client gui on master is not using python-opcua but opcua-asyncio. So if any certificate handling has to be added it is probably inside opcua-asyncio
* Opcua-widgets is about displaying data and adding nodes so i do not think you need to modify anything there
* If you need to add some buttons there is already a "connection settings panel there". It already support loading certificates I think but it might be really ruff. Improvements welcome. Also maybe user and password should be added.
* Also handling of the different connections offered by server should be improved
* And fix all the small bugs around

oroulet, 2021-02-20:
Which commit do you want to modify?

derme302, 2021-02-20:
> Which commit do you want to modify?

In most recent commit on master branch, I was hoping you could amend the commit message to remove the co-author - I must have not had my git settings correct.

Thanks for the feedback, that helps, I'll try putting a couple PRs together and see how how I go.

oroulet, 2021-02-20:
@derme302 I force push the last commit. OK now?

derme302, 2021-02-20:
> @derme302 I force push the last commit. OK now?

Amazing, it looks good, thanks!

closed 2021-02-20 by derme302