FreeOpcUa/opcua-client-gui #62: Run with: opcua-client Problem

fine-tec, 2021-04-13:
Hello, Everyone!

1. I installed PyQT 5

2. I installed python3.6.9 with python-pip3

3. I entered pip3 install opcua-client

4. I couldn't run this problem

Traceback (most recent call last):

  File "/snap/opcua-client/18/bin/opcua-client", line 33, in <module>
    sys.exit(load_entry_point('opcua-client==0.8.0', 'console_scripts', 'opcua-client')())

  File "/snap/opcua-client/18/lib/python3.6/site-packages/pkg_resources/__init__.py", line 474, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)

  File "/snap/opcua-client/18/lib/python3.6/site-packages/pkg_resources/__init__.py", line 2846, in load_entry_point
    return ep.load()

  File "/snap/opcua-client/18/lib/python3.6/site-packages/pkg_resources/__init__.py", line 2450, in load
    return self.resolve()

  File "/snap/opcua-client/18/lib/python3.6/site-packages/pkg_resources/__init__.py", line 2456, in resolve
    module = import(self.module_name, fromlist=['name'], level=0)

  File "/snap/opcua-client/18/lib/python3.6/site-packages/uaclient/mainwindow.py", line 13, in <module>
    from uaclient.theme import breeze_resources
ModuleNotFoundError: No module named 'uaclient.theme'

5. Could you tell me what is wrong?

Thanks for you kindness help!

CuriousCrook, 2021-09-24:
Hello fine-tec,

I have the same problem here. The cause is that the setuptools by default do not install subpackages. With a small change in **setup.py** you can fix the problem.

Change line 1 from
`from setuptools import setup`
to
`from setuptools import setup, find_packages`

and line 8 from
`packages=["uaclient"],`
to
`packages=find_packages(),`

Then you call clone the project and install via pip.

With kind regards
CC

CuriousCrook, 2021-09-29:
I found out a funny thing yesterday. When you install the opcua-client-gui in Linux the sub-package "theme" is installed automatically. But not if you use Windows!

I installed it both times on the same way. 
1. clone
2. pip install -e

If you add the change described in my last comment, it will work in Windows too.

With kind regards
CC