FreeOpcUa/opcua-client-gui #64: Problem with connection

Ilia-urb, 2021-06-02:
asyncua.client.client - WARNING - Deprecated since spec 1.04, call load_data_type_definitions')
asyncua.client.client - WARNING - Deprecated since spec 1.04, call load_data_type_definitions')
uawidgets.utils - ERROR - 'SyncNode' object has no attribute 'get_attributes'')
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/uawidgets/utils.py", line 21, in wrapper
    result = func(self, *args)
  File "/Users/a12313/Downloads/opcua-client-gui-master-2/uaclient/mainwindow.py", line 349, in connect
    self.tree_ui.set_root_node(self.uaclient.client.nodes.root)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/uawidgets/tree_widget.py", line 43, in set_root_node
    self.model.set_root_node(node)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/uawidgets/tree_widget.py", line 167, in set_root_node
    desc = self._get_node_desc(node)
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/uawidgets/tree_widget.py", line 171, in _get_node_desc
    attrs = node.get_attributes([ua.AttributeIds.DisplayName, ua.AttributeIds.BrowseName, ua.AttributeIds.NodeId, ua.AttributeIds.NodeClass])
AttributeError: 'SyncNode' object has no attribute 'get_attributes'
uaclient.mainwindow - WARNING - showing error: %s')

bluet, 2021-06-22:
Same problem here

oroulet, 2021-06-22:
how did you install modeller? I see the name is wrong.. it should be read_attributes

oroulet, 2021-06-22:
I pushed a fix to opcua-modeller master jus tnow

oroulet, 2021-06-23:
ops sorry. this was an issue with ua client gui. how did you instrall? The ua client gui should work if you use master for asyncua (clone from github) I need to do a realease very soon

CuriousCrook, 2021-09-24:
Hello @oroulet,

same problem here.
This is the way I installed it:
- I cloned opcua-asyncio and opcua-client-gui.
- I installed both using pip.
- I have fixed the error described in issue #62 .
- I tried to start the application using a small main programm:

```
from uaclient import mainwindow

if __name__ == '__main__':
    mainwindow.main()
```

With kind regards
CC

kharmabum, 2021-12-06:
Anyone have steps to resolve this issue?

kharmabum, 2021-12-06:
I was able to resolve this issue with the following steps: 

Replace the contents of setup.py with the following :

```
from setuptools import setup, find_packages

setup(name="opcua-client",
      version="0.8.0",
      description="OPC-UA Client GUI",
      author="Olivier R-D",
      url='https://github.com/FreeOpcUa/opcua-client-gui',
      packages=find_packages(),
      license="GNU General Public License",
      install_requires=[
        "asyncua @ git+https://github.com/FreeOpcUa/opcua-asyncio@02525a632e2e14d700ab2d5ec4a18596406640c4", 
        "opcua-widgets @ git+https://github.com/FreeOpcUa/opcua-widgets@efdbb7ef3e9502b1fc3738ab19f1689b2b7d2fa1", 
        "PyQt5"],
      entry_points={'console_scripts':
                    ['opcua-client = uaclient.mainwindow:main']
                    }
      )
```

Then run the following from the project root:

```
pip uninstall opcua-widgets
pip uninstall asyncua
pip install .
```

oroulet, 2021-12-06:
thank you for the report. There has been an issue in releae script. Everything should work now. Version should be 0,.8.2 on pip now