FreeOpcUa/opcua-client-gui #69: ModuleNotFoundError: No module named 'uaclient.theme'

pbertoni89, 2021-12-06:
First encountered in a bare metal Kubuntu 20.04 installation; then it happened again in an existing conda env (python 3.8.8) and in a brand new conda env (python 3.9.7). My desktop is KDE.

```
    conda clean -a
    conda create -n ua3.9 python=3.9  &&  conda activate ua3.9
    pip3 install opcua-client pyqtgraph numpy
```

Produces

> ...
> Installing collected packages: pycparser, typing-extensions, six, cffi, sortedcontainers, pytz, python-dateutil, cryptography, aiosqlite, aiofiles, PyQt5-sip, PyQt5-Qt5, asyncua, PyQt5, opcua-widgets, numpy, pyqtgraph, opcua-client
Successfully installed PyQt5-5.15.6 PyQt5-Qt5-5.15.2 PyQt5-sip-12.9.0 aiofiles-0.8.0 aiosqlite-0.17.0 asyncua-0.9.92 cffi-1.15.0 cryptography-36.0.0 numpy-1.21.4 opcua-client-0.8.2 opcua-widgets-0.6.0 pycparser-2.21 pyqtgraph-0.12.3 python-dateutil-2.8.2 pytz-2021.3 six-1.16.0 sortedcontainers-2.4.0 typing-extensions-4.0.1

In facts

```
$ opcua-client 
Traceback (most recent call last):
  File "/opt/miniconda/envs/ua3.9/bin/opcua-client", line 5, in <module>
    from uaclient.mainwindow import main
  File "/opt/miniconda/envs/ua3.9/lib/python3.9/site-packages/uaclient/mainwindow.py", line 13, in <module>
    from uaclient.theme import breeze_resources
ModuleNotFoundError: No module named 'uaclient.theme'
```

I'm suspecting "breeze" has something to do with Desktop Breeze theme?

```
$ cd /opt/miniconda/envs/ua3.9/lib/python3.9/site-packages/uaclient/
$ grep -rn breeze
mainwindow.py:13:from uaclient.theme import breeze_resources
Binary file __pycache__/mainwindow.cpython-39.pyc matches
```

oroulet, 2021-12-06:
last version updated today I suppose? I see some files are missing in package... something strange is happening...

oroulet, 2021-12-06:
I made a new release 0.8.3 Does it work?

pbertoni89, 2021-12-06:
Yes I was updating today, 10:00 AM CET. Now with `0.8.3` I confirm everything works and I heartily thank you.

closed 2021-12-06 by pbertoni89