FreeOpcUa/opcua-client-gui #70: cannot reconnect

MansonHua, 2021-12-31:
i use opcua-client in ubuntu20.04.
When i connect to my server than shut down my server and restart it, opcua-client cannot reconnect my server automatically..
I want to now, how to realize auto-reconnect.
<details>
<summary>myserver</summary>

```java
import logging
import asyncio
import sys
import time
from asyncua.ua.uatypes import Int32
sys.path.insert(0, "..")
from asyncua import ua, Server
from asyncua.common.methods import uamethod

QtyPoints = 10

@uamethod
def func(parent, value):
    return value * 2

async def main():
    _logger = logging.getLogger('asyncua')
    # setup our server
    server = Server()
    await server.init()
    server.set_endpoint('opc.tcp://0.0.0.0:4840/freeopcua/server/')

    # setup our own namespace, not really necessary but should as spec
    uri = 'http://examples.freeopcua.github.io'
    idx = await server.register_namespace(uri)

    # populating our address space
    # server.nodes, contains links to very common nodes like objects and root
    myobj = await server.nodes.objects.add_object(idx, 'Monitor')
    myvar = {}
    for i in range(QtyPoints):
        var = await myobj.add_variable(idx, 'var' + str(i), 1 )
        await var.set_writable()
        myvar[i] = var
    countvar = await myobj.add_variable(idx, 'count', "none")
    # myobj.add_variable_type()
    await server.nodes.objects.add_method(ua.NodeId('ServerMethod', 2), ua.QualifiedName('ServerMethod', 2), func, [ua.VariantType.Int64], [ua.VariantType.Int64])
    _logger.info('Starting server!')
    async with server:
        count = 0
        new_val = 0
        while True:
            await asyncio.sleep(5)
            new_val = new_val + 1
            count = count + 1
            for i in range(QtyPoints):
                await myvar[i].write_value(new_val)
            if 0 == (count % 100):
                timeNow = time.time()
                writevalue = int(count*QtyPoints)
                writevalue = str(writevalue)+","+str(timeNow)
                await countvar.write_value(writevalue)

if __name__ == '__main__':

    logging.basicConfig(level=logging.WARNING)

    asyncio.run(main(), debug=False)```
</details>

oroulet, 2022-01-28:
reconnecting is not so easy so it i snot implemented

ondrno, 2023-10-30:
@oroulet : I would be also also interested in the reconnect functionality. Do you have any plans to implement this in the near future?