FreeOpcUa/opcua-client-gui #72: AttributeError: module 'asyncua.ua' has no attribute 'ParameterResultDataType'

AlexanderB-C, 2022-01-14:
Hi,

I can successfully connect to the uaserver.exe demo server of the opcua-asyncio package using this package (opcua-client-gui).

However I am trying to connect to a Siemens OPC-UA server now which fails with this output in console:

```
pyqtgraph or numpy are not installed, use of graph feature disabled
ADR ['opc.tcp://localhost:4840', 'opc.tcp://localhost:53530/OPCUA/SimulationServer/']
uaclient.uaclient - INFO - Connecting to opc.tcp://192.168.0.1:4840 with parameters None, None, None, None')
asyncua.client.client - WARNING - Requested session timeout to be 3600000ms, got 30000ms instead')
QObject::connect: Cannot queue arguments of type 'QTextCursor'
(Make sure 'QTextCursor' is registered using qRegisterMetaType().)
asyncua.common.structures104 - ERROR - None_ - EnumField(Value=0, DisplayName=LocalizedText(Locale=None, Text='None'), Description=LocalizedText(Locale=None, Text=None), Name='None') False

class TraceLevel(IntEnum):

    '''
    TraceLevel EnumInt autogenerated from EnumDefinition
    '''

    Error = 32
    Warning = 48
    System = 56
    Info = 60
    Debug = 62
    Content = 63
    All = -1
    None_ = 0
')
asyncua.common.structures104 - ERROR - PgmTest - EnumField(Value=19, DisplayName=LocalizedText(Locale=None, Text='PgmTest'), Description=LocalizedText(Locale=None, Text=None), Name='PgmTest') False

class SimaticOperatingState(IntEnum):

    '''
    SimaticOperatingState EnumInt autogenerated from EnumDefinition
    '''

    NotSupported = 0
    StopFwUpdate = 1
    StopSelfInitialization = 3
    Stop = 4
    Startup = 6
    Run = 8
    RunRedundant = 9
    Halt = 10
    RunSyncUp = 11
    SyncUp = 12
    Defective = 13
    ErrorSearch = 14
    NoPower = 15
    CiR = 16
    STOPwithoutODIS = 17
    RunODIS = 18
    PgmTest = 19
')
asyncua.client.client - WARNING - Deprecated since spec 1.04, call load_data_type_definitions')
asyncua.client.client - WARNING - Deprecated since spec 1.04, call load_data_type_definitions')

@dataclass
class FetchResultDataType:

    '''
    FetchResultDataType structure autogenerated from xml
    '''

@dataclass
class TransferResultErrorDataType:

    '''
    TransferResultErrorDataType structure autogenerated from xml
    '''

    Status:ua.Int32 = 0
    Diagnostics:ua.DiagnosticInfo = ua.DiagnosticInfo()

@dataclass
class TransferResultDataDataType:

    '''
    TransferResultDataDataType structure autogenerated from xml
    '''

    SequenceNumber:ua.Int32 = 0
    EndOfResults:ua.Boolean = True
    ParameterDefs:List[ua.ParameterResultDataType] = field(default_factory=list)

asyncua.common.structures - ERROR - Failed to execute auto-generated code from UA datatype:

@dataclass
class TransferResultDataDataType:

    '''
    TransferResultDataDataType structure autogenerated from xml
    '''

    SequenceNumber:ua.Int32 = 0
    EndOfResults:ua.Boolean = True
    ParameterDefs:List[ua.ParameterResultDataType] = field(default_factory=list)
')
Traceback (most recent call last):
  File "C:\Python39\lib\site-packages\asyncua\common\structures.py", line 281, in _generate_python_class
    exec(code, env)
  File "<string>", line 4, in <module>
  File "<string>", line 12, in TransferResultDataDataType
AttributeError: module 'asyncua.ua' has no attribute 'ParameterResultDataType'
uaclient.mainwindow - WARNING - showing error: %s')
uawidgets.utils - ERROR - module 'asyncua.ua' has no attribute 'ParameterResultDataType'')
Traceback (most recent call last):
  File "C:\Python39\lib\site-packages\uawidgets\utils.py", line 21, in wrapper
    result = func(self, *args)
  File "C:\Python39\lib\site-packages\uaclient\mainwindow.py", line 343, in connect
    self.uaclient.connect(uri)
  File "C:\Python39\lib\site-packages\uaclient\uaclient.py", line 96, in connect
    self.client.load_type_definitions()
  File "C:\Python39\lib\site-packages\asyncua\sync.py", line 94, in wrapper
    result = self.tloop.post(aio_func(*args, **kwargs))
  File "C:\Python39\lib\site-packages\asyncua\sync.py", line 52, in post
    return futur.result()
  File "C:\Python39\lib\concurrent\futures\_base.py", line 445, in result
    return self.__get_result()
  File "C:\Python39\lib\concurrent\futures\_base.py", line 390, in __get_result
    raise self._exception
  File "C:\Python39\lib\site-packages\asyncua\client\client.py", line 668, in load_type_definitions
    return await load_type_definitions(self, nodes)
  File "C:\Python39\lib\site-packages\asyncua\common\structures.py", line 223, in load_type_definitions
    generator.get_python_classes(structs_dict)
  File "C:\Python39\lib\site-packages\asyncua\common\structures.py", line 177, in get_python_classes
    return _generate_python_class(self.model, env=env)
  File "C:\Python39\lib\site-packages\asyncua\common\structures.py", line 281, in _generate_python_class
    exec(code, env)
  File "<string>", line 4, in <module>
  File "<string>", line 12, in TransferResultDataDataType
AttributeError: module 'asyncua.ua' has no attribute 'ParameterResultDataType'
uaclient.mainwindow - WARNING - showing error: %s')
```

Am I doing something wrong or is this a bug on package side?

Thanks & best regards
Alexander

deeTEEcee, 2022-01-20:
im getting the same error

oroulet, 2022-01-28:
I pushed a fix today to not crash in case of that kind of erros. It might help