FreeOpcUa/opcua-client-gui #73: Problems to navigate in the opcua-client-gui

3D-Schorsch, 2022-01-27:
When I press tab in the GUI in the OPCUA Free Client from you I end up at the bottom of the log at some point. So far no problem. But if I keep pressing Tab I can't get out of the logging window. Instead I write more tabs into the logging console and can't get out with tab anymore. 

Otherwise I find the GUI very well operable.

fcnjd, 2022-01-28:
Hi,
thank you for bringing this up, I totally agree. Especially for blind people using a screenreader and only the keyboard, like me, this is very difficult, because once you tap in, you're basically trapped since you cannot tap out again, and so have to restart the whole client. From a user's perspective, I'll propose the two following solutions I can think of:
* Either you remove the possibility to insert the tab sign, making the Tab key behalf as usual and move the focus to the next focusable element
* Or, you make it possible to use a common key combination to get out of that again (I'd suggest strg+tab (and then also strg+shift+tab), or F6, or even both possibilities), and then somehow indicate to the user that s/he has to press this key to get out, for example by writing this into the log or making a popup indicating that.
* I'd be very happy to see a fix for that.

oroulet, 2022-01-28:
I think have not looked at these UI thing for many years, but I just pushed a commit to solve your issue https://github.com/FreeOpcUa/opcua-client-gui/commit/add894076d4fff28ccdcca475862ac3f93c1b6aa

does that solve it for you ?

fcnjd, 2022-01-30:
@oroulet Thank you very much for your fix! Just looked at it, and it works like a charm to tab over the logs.

fcnjd, 2022-02-19:
@3D-Schorsch Since this is fixed now, do you wanna close this?

closed 2022-02-20 by oroulet