FreeOpcUa/opcua-client-gui #74: No way to get back from subscription view by keyboard

fcnjd, 2022-02-19:
As a blind user who only uses the keyboard to navigate, I want to be able to navigate back from the subscriptions view to the main window with the connection and attributes view. Currently, when I subscribe to one or more data changes using the context menu, and then use tab to get to the subscriptions table, there's no way to get back again from it. I would be very happy if you could find a fix for that.

oroulet, 2022-02-20:
Never thought at that kind of thigns. but I fixed that on master. Also fixed the order of windows using Tab

fcnjd, 2023-09-19:
Hey,
thank you for your efforts in that, and sorry for leaving this open and uncommented so long. The problems are now fixed, so I can close this issue.

closed 2023-09-19 by fcnjd