FreeOpcUa/opcua-client-gui #76: RuntimeWarning: coroutine 'MonitoredItemService.datachange_callback' was never awaited

francescobianca, 2022-02-20:
Hi,

I am trying to use the following library function:

`await node.delete(True, True)`

where the selected node is the root node.
This is the data model:
```
{
  "Id": "Site1",
  .....
}
```

[image]

and the selected node to delete is **Site1**
the output of 

```
node_name = await node.read_display_name()
logger.debug(f"{node_name.Text}")
```

is _Site1_. (as proof)

The deletion of the object works but I don't understand from what is originated the warning.

**Version**<br />
Python-Version: 3.8 <br />
opcua-asyncio Version 0.9.92

oroulet, 2022-02-20:
where do you see that warning? are you using the UI or writting code?

francescobianca, 2022-02-21:
Oh yes, sorry is a problem found in the opcua-asyncio library. I have a server on which I try to call the deletion of a node but I get that warning.