FreeOpcUa/opcua-client-gui #80: Unknown datatype for field

jmtatsch, 2022-05-17:
Hi, 

when I initially connect to a Siemens S7 it dies due to this:

```
uawidgets.utils - ERROR - Unknown datatype for field: StructureField(Name='awInt', Description=LocalizedText(Locale=None, Text=None), DataType=NodeId(Identifier=3002, NamespaceIndex=3, NodeIdType=<NodeIdType.FourByte: 1>), ValueRank=1, ArrayDimensions=[3], MaxStringLength=0, IsOptional=False) in structure:_LCB_15x_ActuatorSpeed_B_1_scInternalCtrl_, please report')
Traceback (most recent call last):
  File "/home/user/.local/lib/python3.8/site-packages/uawidgets/utils.py", line 21, in wrapper
    result = func(self, *args)
  File "/home/user/.local/lib/python3.8/site-packages/uaclient/mainwindow.py", line 354, in connect
    self.uaclient.connect(uri)
  File "/home/user/.local/lib/python3.8/site-packages/uaclient/uaclient.py", line 126, in connect
    self.client.load_data_type_definitions()
  File "/home/user/.local/lib/python3.8/site-packages/asyncua/sync.py", line 94, in wrapper
    result = self.tloop.post(aio_func(*args, **kwargs))
  File "/home/user/.local/lib/python3.8/site-packages/asyncua/sync.py", line 52, in post
    return futur.result()
  File "/usr/lib/python3.8/concurrent/futures/_base.py", line 444, in result
    return self.__get_result()
  File "/usr/lib/python3.8/concurrent/futures/_base.py", line 389, in __get_result
    raise self._exception
  File "/home/user/.local/lib/python3.8/site-packages/asyncua/client/client.py", line 676, in load_data_type_definitions
    return await load_data_type_definitions(self, node, overwrite_existing=overwrite_existing)
  File "/home/user/.local/lib/python3.8/site-packages/asyncua/common/structures104.py", line 324, in load_data_type_definitions
    env = await _generate_object(dts.name, dts.sdef, data_type=dts.data_type)
  File "/home/user/.local/lib/python3.8/site-packages/asyncua/common/structures104.py", line 242, in _generate_object
    code = make_structure_code(data_type, name, sdef)
  File "/home/user/.local/lib/python3.8/site-packages/asyncua/common/structures104.py", line 192, in make_structure_code
    raise RuntimeError(f"Unknown datatype for field: {sfield} in structure:{struct_name}, please report")
```

Any idea what goes wrong?

slokhorst, 2022-08-02:
I'm having the same problem when connecting to a PLC that is serving a node with the `STRING` data type:
[image]
```
RuntimeError: Unknown datatype for field: StructureField(Name='tijdvak', Description=LocalizedText(Locale=None, Text=''), DataType=NodeId(Identifier=3014, NamespaceIndex=3, NodeIdType=<NodeIdType.FourByte: 1>), ValueRank=-1, ArrayDimensions=[], MaxStringLength=0, IsOptional=False) in structure:CounterEntry, please report
uaclient.mainwindow - WARNING - showing error: %s')
```

schroeder-, 2022-08-03:
I patch some errors with custom datatypes, but there wasn't any new version released yet. So you could try to install the current master of [opcua-asyncio](https://github.com/FreeOpcUa/opcua-asyncio).

jmtatsch, 2022-08-08:
Was fixed with https://github.com/FreeOpcUa/opcua-asyncio/pull/901

closed 2022-08-08 by jmtatsch