FreeOpcUa/opcua-client-gui #82: connect OPCUA Server ERROR：raise RuntimeError(f"Unknown datatype for field: {sfield} in structure:{struct_name}, please report")

duchuan34, 2022-06-02:
ERROR：raise RuntimeError(f"Unknown datatype for field: {sfield} in structure:{struct_name}, please report")

duchuan34, 2022-06-02:
@oroulet  @CuriousCrook  what datatype from log，please guid me how to connect。

CuriousCrook, 2022-06-02:
I once had a similar error, but didn't have time to investigate it further at the time. At that time it seemed that the client would not support custom structures within structures. 

Do you have the possibility to reproduce the problem with the Python server? If you have a NodeSet2.xml for the problematic types, you could load it in a sample server and connect the GUI client to it. If the error is then reproducible, that would be super handy for us to investigate the problem further.

duchuan34, 2022-06-02:
@CuriousCrook thank U for reply，uaexpert can connect UA Server success by this models。
models XML INFO，this datetypes include on opcua C/C++，  my server by C/C++ SDK dev。
OS：Win 10
Interl5/X64

<?xml version="1.0" encoding="UTF-8"?>
<UANodeSet xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:uax="http://opcfoundation.org/UA/2008/02/Types.xsd" xmlns:s1="http://unified-automation.com/CppDemoServer/BuildingAutomationXml" xmlns:s2="urn:UnifiedAutomation:CppDemoServer:BuildingAutomation" xmlns:xsd="http://www.w3.org/2001/XMLSchema" xmlns="http://opcfoundation.org/UA/2011/03/UANodeSet.xsd" Version="v1" LastModified="Fri May 20 13:50:8 2022">
	<NamespaceUris>
		<Uri>urn:inovance:rtdb:ns-1001</Uri>
	</NamespaceUris>
	<Aliases>
		<Alias Alias="Boolean">i=1</Alias>
		<Alias Alias="SByte">i=2</Alias>
		<Alias Alias="Byte">i=3</Alias>
		<Alias Alias="Int16">i=4</Alias>
		<Alias Alias="UInt16">i=5</Alias>
		<Alias Alias="Int32">i=6</Alias>
		<Alias Alias="UInt32">i=7</Alias>
		<Alias Alias="Int64">i=8</Alias>
		<Alias Alias="UInt64">i=9</Alias>
		<Alias Alias="Float">i=10</Alias>
		<Alias Alias="Double">i=11</Alias>
		<Alias Alias="String">i=12</Alias>
		<Alias Alias="DateTime">i=13</Alias>
		<Alias Alias="Guid">i=14</Alias>
		<Alias Alias="ByteString">i=15</Alias>
		<Alias Alias="XmlElement">i=16</Alias>
		<Alias Alias="NodeId">i=17</Alias>
		<Alias Alias="ExpandedNodeId">i=18</Alias>
		<Alias Alias="StatusCode">i=19</Alias>
		<Alias Alias="QualifiedName">i=20</Alias>
		<Alias Alias="LocalizedText">i=21</Alias>
		<Alias Alias="Structure">i=22</Alias>
		<Alias Alias="DataValue">i=23</Alias>
		<Alias Alias="BaseDataType">i=24</Alias>
		<Alias Alias="DiagnosticInfo">i=25</Alias>
		<Alias Alias="Number">i=26</Alias>
		<Alias Alias="Integer">i=27</Alias>
		<Alias Alias="UInterger">i=28</Alias>
		<Alias Alias="Enumeration">i=29</Alias>
		<Alias Alias="Image">i=30</Alias>
		<Alias Alias="Organizes">i=35</Alias>
		<Alias Alias="HasModellingRule">i=37</Alias>
		<Alias Alias="HasTypeDefinition">i=40</Alias>
		<Alias Alias="HasSubtype">i=45</Alias>
		<Alias Alias="HasProperty">i=46</Alias>
		<Alias Alias="HasComponent">i=47</Alias>
		<Alias Alias="UtcTime">i=294</Alias>
		<Alias Alias="Argument">i=296</Alias>
		<Alias Alias="Range">i=884</Alias>
		<Alias Alias="EUInformation">i=887</Alias>
		<Alias Alias="ImageBMP">i=2000</Alias>
		<Alias Alias="ImageGIF">i=2001</Alias>
		<Alias Alias="ImageJPG">i=2002</Alias>
		<Alias Alias="ImagePNG">i=2003</Alias>
		<Alias Alias="TimeZoneDataType">i=8912</Alias>
	</Aliases>
	<UAObject NodeId="ns=1001;s=varGroup0" BrowseName="varGroup0">
		<DisplayName>varGroup0</DisplayName>
		<Description/>
		<References>
			<Reference ReferenceType="HasTypeDefinition">ns=10;i=61</Reference>
			<Reference ReferenceType="Organizes" IsForward="false">i=85</Reference>
		</References>
	</UAObject>
	<UAVariable DataType="Boolean" NodeId="ns=1001;s=TEST_EXTERNAL_G0_L1_1_0" BrowseName="TEST_EXTERNAL_G0_L1_1_0" UserAccessLevel="3" AccessLevel="3">
		<DisplayName>TEST_EXTERNAL_G0_L1_1_0</DisplayName>
		<Description>描述信息</Description>
		<References>
			<Reference ReferenceType="HasTypeDefinition">ns=10;i=2002</Reference>
			<Reference ReferenceType="HasComponent" IsForward="false">ns=1001;s=varGroup0</Reference>
		</References>
		<Value>
			<uax:Boolean/>
		</Value>
	</UAVariable>
	<UAVariable ParentNodeId="ns=1001;s=TEST_EXTERNAL_G0_L1_1_0" NodeId="ns=1001;s=TEST_EXTERNAL_G0_L1_1_0.VariableCfgs" BrowseName="VariableCfgs">
		<DisplayName>VariableCfgs</DisplayName>
		<References>
			<Reference ReferenceType="HasTypeDefinition">ns=10;i=3001</Reference>
		</References>
		<Extensions>
			<Extension>
				<ConfigInfos type="2">
					<ItemInfos Name="values" DataType="6" Array="true">
						<Value>10</Value>
						<Value>20</Value>
					</ItemInfos>
					<ItemInfos Name="definitions" DataType="12" Array="true">
						<Value>Description1</Value>
						<Value>Description2</Value>
					</ItemInfos>
				</ConfigInfos>
			</Extension>
			<Extension>
				<ConfigInfos type="3">
					<ItemInfos Name="values" DataType="6" Array="true">
						<Value>0</Value>
						<Value>1</Value>
					</ItemInfos>
					<ItemInfos Name="messages" DataType="12" Array="true">
						<Value>message 0</Value>
						<Value>message 1</Value>
					</ItemInfos>
				</ConfigInfos>
			</Extension>
			<Extension>
				<ConfigInfos type="4">
					<ItemInfos Name="compares" DataType="5" Array="false">
						<Value>TEST_EXTERNAL_G0_L1_1_0</Value>
					</ItemInfos>
					<ItemInfos Name="values" DataType="11" Array="false">
						<Value>TEST_EXTERNAL_G0_L1_1_0</Value>
					</ItemInfos>
					<ItemInfos Name="trends" DataType="5" Array="false">
						<Value>TEST_EXTERNAL_G0_L1_1_0</Value>
					</ItemInfos>
					<ItemInfos Name="scripts" DataType="12" Array="false">
						<Value>TEST_EXTERNAL_G0_L1_1_0</Value>
					</ItemInfos>
				</ConfigInfos>
			</Extension>
		</Extensions>
	</UAVariable>
	<UAVariable DataType="Boolean" ParentNodeId="ns=1001;s=TEST_EXTERNAL_G0_L1_1_0" NodeId="ns=1001;s=TEST_EXTERNAL_G0_L1_1_0.label" BrowseName="TEST_EXTERNAL_G0_L1_1_0.label" UserAccessLevel="3" AccessLevel="3">
		<DisplayName>TEST_EXTERNAL_G0_L1_1_0.label</DisplayName>
		<References>
			<Reference ReferenceType="HasTypeDefinition">ns=10;i=2005</Reference>
			<Reference ReferenceType="HasComponent" IsForward="false">ns=20;s=links1</Reference>
			<Reference ReferenceType="HasComponent" IsForward="false">ns=1001;s=TEST_EXTERNAL_G0_L1_1_0</Reference>
		</References>
	</UAVariable>
	<UAVariable ParentNodeId="ns=1001;s=TEST_EXTERNAL_G0_L1_1_0" NodeId="ns=1001;s=TEST_EXTERNAL_G0_L1_1_0.AlarmCfgs" BrowseName="AlarmCfgs">
		<DisplayName>AlarmCfgs</DisplayName>
		<References>
			<Reference ReferenceType="HasTypeDefinition">ns=10;i=3002</Reference>
		</References>
	</UAVariable>
</UANodeSet>

schroeder-, 2022-06-02:
I fix a lot of problems with structs in this pull request: https://github.com/FreeOpcUa/opcua-asyncio/pull/901 maybe you can test it.

duchuan34, 2022-06-06:
> I fix a lot of problems with structs in this pull request: [FreeOpcUa/opcua-asyncio#901](https://github.com/FreeOpcUa/opcua-asyncio/pull/901) maybe you can test it.
asyncua/ua/uatypes.py file is 最新的 
asyncua/common/structures104.py allpath is C:\Users\Administrator\AppData\Local\Programs\Python\Python37\Lib\site-packages\asyncua\common\structures104.py, I made a replacement。

python 3.7.0rc1
OS win10 64
48010 server ：C++ OPCua
opcua-asyncio：0.9.94

[image]

schroeder-, 2022-06-06:
The fix is not released yet. You must clone asyncua from current master and install it via 
`pip install .`

duchuan34, 2022-06-07:
e/opcuapythonTest/opcua-client-gui
 
 507  pip uninstall  asyncua opcua-client
  508  pip install .
  510  opcua-client
same err：
uaclient.uaclient - INFO - Connecting to opc.tcp://localhost:48010 with parameters None, None, None, None')
uaclient.mainwindow - WARNING - showing error: %s')
uawidgets.utils - ERROR - Unknown datatype for field: StructureField(Name='Value', Description=LocalizedText(Locale=None, Text=None), DataType=NodeId(Identifier=40016, NamespaceIndex=10, NodeIdType=<NodeIdType.FourByte: 1>), ValueRank=-1, ArrayDimensions=[], MaxStringLength=0, IsOptional=False) in structure:MessageData, please report')
Traceback (most recent call last):
  File "c:\users\administrator\appdata\local\programs\python\python37\lib\site-packages\uawidgets\utils.py", line 21, in wrapper
    result = func(self, *args)
  File "c:\users\administrator\appdata\local\programs\python\python37\lib\site-packages\uaclient\mainwindow.py", line 354, in connect
    self.uaclient.connect(uri)
  File "c:\users\administrator\appdata\local\programs\python\python37\lib\site-packages\uaclient\uaclient.py", line 126, in connect
    self.client.load_data_type_definitions()
  File "c:\users\administrator\appdata\local\programs\python\python37\lib\site-packages\asyncua\sync.py", line 94, in wrapper
    result = self.tloop.post(aio_func(*args, **kwargs))
  File "c:\users\administrator\appdata\local\programs\python\python37\lib\site-packages\asyncua\sync.py", line 52, in post
    return futur.result()
  File "c:\users\administrator\appdata\local\programs\python\python37\lib\concurrent\futures\_base.py", line 432, in result
    return self.__get_result()
  File "c:\users\administrator\appdata\local\programs\python\python37\lib\concurrent\futures\_base.py", line 384, in __get_result
    raise self._exception
  File "c:\users\administrator\appdata\local\programs\python\python37\lib\site-packages\asyncua\client\client.py", line 685, in load_data_type_definitions
    return await load_data_type_definitions(self, node, overwrite_existing=overwrite_existing)
  File "c:\users\administrator\appdata\local\programs\python\python37\lib\site-packages\asyncua\common\structures104.py", line 356, in load_data_type_definitions
    env = await _generate_object(dts.name, dts.sdef, data_type=dts.data_type)
  File "c:\users\administrator\appdata\local\programs\python\python37\lib\site-packages\asyncua\common\structures104.py", line 274, in _generate_object
    code = make_structure_code(data_type, name, sdef)
  File "c:\users\administrator\appdata\local\programs\python\python37\lib\site-packages\asyncua\common\structures104.py", line 199, in make_structure_code
    raise RuntimeError(f"Unknown datatype for field: {sfield} in structure:{struct_name}, please report")
RuntimeError: Unknown datatype for field: StructureField(Name='Value', Description=LocalizedText(Locale=None, Text=None), DataType=NodeId(Identifier=40016, NamespaceIndex=10, NodeIdType=<NodeIdType.FourByte: 1>), ValueRank=-1, ArrayDimensions=[], MaxStringLength=0, IsOptional=False) in structure:MessageData, please report
uaclient.mainwindow - WARNING - showing error: %s')

@schroeder

schroeder-, 2022-06-07:
You don't need to update opcua-client, you need to install the current master from asyncio: https://github.com/FreeOpcUa/opcua-asyncio.git

stan1334, 2022-06-07:
> You don't need to update opcua-client, you need to install the current master from asyncio: https://github.com/FreeOpcUa/opcua-asyncio.git

realy realy cannot connect，python not support C/CXX some base type

closed 2023-01-14 by stan1334