FreeOpcUa/opcua-client-gui #87: AttributeError: module 'asyncua.ua' has no attribute 'Integer'

amattana, 2022-08-23:
Just installed with pip.
Trying to connect to a Rittal PDU with OPC-UA server running.
Query server capabilities works.
Any hints?

schroeder-, 2022-08-23:
A minimal code, stacktrace and log output would be nice to help you.

schroeder-, 2022-08-23:
Ok found the problem, I will create a pr to fix the issue soon.

amattana, 2022-08-23:
Great, thanks!

schroeder-, 2022-08-23:
Can you install this branch: https://github.com/FreeOpcUa/opcua-asyncio/pull/1009 and retry?

amattana, 2022-08-23:
Got the same error, maybe I'm wrong...!?!

$ pip install git+https://github.com/schroeder-/opcua-asyncio.git
Collecting git+https://github.com/schroeder-/opcua-asyncio.git
  Cloning https://github.com/schroeder-/opcua-asyncio.git to /tmp/pip-req-build-8oulfcks
  Running command git clone --filter=blob:none --quiet https://github.com/schroeder-/opcua-asyncio.git /tmp/pip-req-build-8oulfcks
  Resolved https://github.com/schroeder-/opcua-asyncio.git to commit c6596b0cc4a881ff16e92f2e3cf9de58c349fe3d
  Preparing metadata (setup.py) ... done
Requirement already satisfied: aiofiles in ./Software/aavs-system/py3venv/python/lib/python3.8/site-packages (from asyncua==0.9.94) (0.8.0)
Requirement already satisfied: aiosqlite in ./Software/aavs-system/py3venv/python/lib/python3.8/site-packages (from asyncua==0.9.94) (0.17.0)
Requirement already satisfied: python-dateutil in ./Software/aavs-system/py3venv/python/lib/python3.8/site-packages (from asyncua==0.9.94) (2.8.2)
Requirement already satisfied: pytz in ./Software/aavs-system/py3venv/python/lib/python3.8/site-packages (from asyncua==0.9.94) (2022.1)
Requirement already satisfied: cryptography in ./Software/aavs-system/py3venv/python/lib/python3.8/site-packages (from asyncua==0.9.94) (37.0.4)
Requirement already satisfied: sortedcontainers in ./Software/aavs-system/py3venv/python/lib/python3.8/site-packages (from asyncua==0.9.94) (2.4.0)
Requirement already satisfied: typing_extensions>=3.7.2 in ./Software/aavs-system/py3venv/python/lib/python3.8/site-packages (from aiosqlite->asyncua==0.9.94) (4.1.1)
Requirement already satisfied: cffi>=1.12 in ./Software/aavs-system/py3venv/python/lib/python3.8/site-packages (from cryptography->asyncua==0.9.94) (1.15.1)
Requirement already satisfied: six>=1.5 in ./Software/aavs-system/py3venv/python/lib/python3.8/site-packages (from python-dateutil->asyncua==0.9.94) (1.16.0)
Requirement already satisfied: pycparser in ./Software/aavs-system/py3venv/python/lib/python3.8/site-packages (from cffi>=1.12->cryptography->asyncua==0.9.94) (2.21)

$ opcua-client
ADR ['opc.tcp://localhost:4840', 'opc.tcp://localhost:53530/OPCUA/SimulationServer/']
uaclient.uaclient - INFO - Endpoint 1:')
uaclient.uaclient - INFO -   Endpoint URL: opc.tcp://192.168.30.41:4840')
uaclient.uaclient - INFO -   Application URI: urn:eth0:Rittal:CMC_III_UA_Server')
uaclient.uaclient - INFO -   Product URI: http://www.rittal.com')
uaclient.uaclient - INFO -   Application Name: LocalizedText(Locale='', Text='CMC III UA Server')')
uaclient.uaclient - INFO -   Application Type: ApplicationType.Server')
uaclient.uaclient - INFO -   Discovery URL: opc.tcp://eth0:4840')
uaclient.uaclient - INFO -   Server Certificate: domainComponent=eth0, countryName=DE, stateOrProvinceName=Hessen, localityName=Herborn, organizationName=Rittal GmbH, commonName=CMCIII (self-signed), 2022-08-23 10:50:42 - 2027-08-22 10:50:42')
uaclient.uaclient - INFO -   Security Mode: MessageSecurityMode.None_')
uaclient.uaclient - INFO -   Security Policy URI: http://opcfoundation.org/UA/SecurityPolicy#None')
uaclient.uaclient - INFO -   User policy: 0')
uaclient.uaclient - INFO -     Token type: UserTokenType.Anonymous')
uaclient.uaclient - INFO -   User policy: 1')
uaclient.uaclient - INFO -     Token type: UserTokenType.UserName')
uaclient.uaclient - INFO -     Security Policy URI: http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15')
uaclient.uaclient - INFO -   User policy: 2')
uaclient.uaclient - INFO -     Token type: UserTokenType.Certificate')
uaclient.uaclient - INFO -     Security Policy URI: http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15')
uaclient.uaclient - INFO -   Transport Profile URI: http://opcfoundation.org/UA-Profile/Transport/uatcp-uasc-uabinary')
uaclient.uaclient - INFO -   Security Level: 0')
uaclient.uaclient - INFO - ')
uaclient.uaclient - INFO - Endpoint 2:')
uaclient.uaclient - INFO -   Endpoint URL: opc.tcp://192.168.30.41:4840')
uaclient.uaclient - INFO -   Application URI: urn:eth0:Rittal:CMC_III_UA_Server')
uaclient.uaclient - INFO -   Product URI: http://www.rittal.com')
uaclient.uaclient - INFO -   Application Name: LocalizedText(Locale='', Text='CMC III UA Server')')
uaclient.uaclient - INFO -   Application Type: ApplicationType.Server')
uaclient.uaclient - INFO -   Discovery URL: opc.tcp://eth0:4840')
uaclient.uaclient - INFO -   Server Certificate: domainComponent=eth0, countryName=DE, stateOrProvinceName=Hessen, localityName=Herborn, organizationName=Rittal GmbH, commonName=CMCIII (self-signed), 2022-08-23 10:50:42 - 2027-08-22 10:50:42')
uaclient.uaclient - INFO -   Security Mode: MessageSecurityMode.SignAndEncrypt')
uaclient.uaclient - INFO -   Security Policy URI: http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15')
uaclient.uaclient - INFO -   User policy: 0')
uaclient.uaclient - INFO -     Token type: UserTokenType.Anonymous')
uaclient.uaclient - INFO -   User policy: 1')
uaclient.uaclient - INFO -     Token type: UserTokenType.UserName')
uaclient.uaclient - INFO -     Security Policy URI: http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15')
uaclient.uaclient - INFO -   User policy: 2')
uaclient.uaclient - INFO -     Token type: UserTokenType.Certificate')
uaclient.uaclient - INFO -     Security Policy URI: http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15')
uaclient.uaclient - INFO -   Transport Profile URI: http://opcfoundation.org/UA-Profile/Transport/uatcp-uasc-uabinary')
[... cut ...]
  File "/home/aavs/Software/aavs-system/py3venv/python/lib/python3.8/site-packages/asyncua/common/structures104.py", line 376, in _recursive_parse_basedatatypes
    env = make_basetype_code(name, parent_datatype)
  File "/home/aavs/Software/aavs-system/py3venv/python/lib/python3.8/site-packages/asyncua/common/structures104.py", line 394, in make_basetype_code
    exec(code, env)
  File "<string>", line 2, in <module>
AttributeError: module 'asyncua.ua' has no attribute 'Integer'
uaclient.mainwindow - WARNING - showing error: %s')
Disconnecting from server

schroeder-, 2022-08-23:
you install my master branch not the pr try this:

pip install git+https://github.com/schroeder-/opcua-asyncio.git@add_integer_uinteger