FreeOpcUa/opcua-client-gui #90: Auth via username/password funtionality

georges-schutz, 2023-01-05:
First thank you to all for all this wonderful work.

I face an issue for the FlowChief OPCUA server as this currently only accepts user-authentication via username/password combination and not via the user-certificat/key identification. For the application cert/key the server is correctly handling this.

The asyncua Client-Class does support the username/password authentication method and I also tested the connectivity to the FlowChief-OPCUA server using this low level client class and this works fine.
I could try to develop this as an alternative authentication method by adding this to the connection dialog where currently the user-certificat/key can be specified.

Is there anybody else being interested by this functionality?

axi92, 2023-01-25:
#35

georges-schutz, 2023-01-26:
Thank you for that link and yes I must have missed this when I searched for the support of that functionality.
This is solving my basic issue as I can now use the GUI to also connect to FlowChief servers.
But I have a bit of an issue with security using that way as the URI is stored in the history of connected servers and with this the username/password is visible in clear (and usable) for anyone being able to access the system where the Client is hosted.
There should also be a way where you could better hide the credentials also for a simple username/password authentication.

closed 2023-01-26 by georges-schutz