FreeOpcUa/opcua-client-gui #97: Snap Package is broken

sruehl, 2023-08-22:
When running the package from the snapstore following error occurs:
```
❯ opcua-client
Traceback (most recent call last):
  File "/snap/opcua-client/14/bin/opcua-client", line 33, in <module>
    sys.exit(load_entry_point('opcua-client==0.8.0', 'console_scripts', 'opcua-client')())
  File "/snap/opcua-client/14/lib/python3.6/site-packages/pkg_resources/__init__.py", line 473, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/snap/opcua-client/14/lib/python3.6/site-packages/pkg_resources/__init__.py", line 2843, in load_entry_point
    return ep.load()
  File "/snap/opcua-client/14/lib/python3.6/site-packages/pkg_resources/__init__.py", line 2447, in load
    return self.resolve()
  File "/snap/opcua-client/14/lib/python3.6/site-packages/pkg_resources/__init__.py", line 2453, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/snap/opcua-client/14/lib/python3.6/site-packages/uaclient/mainwindow.py", line 12, in <module>
    from asyncua import ua
  File "/snap/opcua-client/14/lib/python3.6/site-packages/asyncua/__init__.py", line 6, in <module>
    from .client import Client
  File "/snap/opcua-client/14/lib/python3.6/site-packages/asyncua/client/__init__.py", line 1, in <module>
    from .client import Client
  File "/snap/opcua-client/14/lib/python3.6/site-packages/asyncua/client/client.py", line 18, in <module>
    from ..crypto import uacrypto, security_policies
  File "/snap/opcua-client/14/lib/python3.6/site-packages/asyncua/crypto/uacrypto.py", line 15, in <module>
    from dataclasses import dataclass
ModuleNotFoundError: No module named 'dataclasses'
```

sruehl, 2023-08-22:
was already reported here but ignored: https://github.com/FreeOpcUa/opcua-client-gui/issues/93

sruehl, 2023-08-22:
snap was installed from https://snapcraft.io/opcua-client

oroulet, 2023-08-22:
That package is outdated. I created it once with one click using some snap/github integration, but never found out how to update it later... and I tried several times and I have a Phd in a supposedly relevant field ;-(

oroulet, 2023-08-22:
so if you know how to do that, you are welcome to try or explain here...

sruehl, 2023-08-22:
No idea, but as a naive approach I would try to setup a action with https://github.com/marketplace/actions/snapcraft-action and limit it to `on: tags`. Do you remember where you exactly struggled or what the problem was?

sruehl, 2023-08-22:
Actually I think it might be better to use the official actions https://github.com/snapcore/action-build and https://github.com/snapcore/action-publish.

sruehl, 2023-08-22:
Here is a bit more complete example https://snapcraft.io/docs/ros-with-github-actions.

I could prepare a PR but can't help debugging if I'm not able to see the run output (might be restricted to crontributer only)

sruehl, 2023-08-22:
so TLDR: you need to to create a Github secret called `STORE_LOGIN` and fill it with the output as described here https://github.com/snapcore/action-publish#store-login.

Once that is done you can merge the PR and we check from there what needs to be done.